nicolapiccinelli/libmpc #3: test_lmpc fails with static assertion OBJECT_ALLOCATED_ON_STACK_IS_TOO_BIG

vkotaru, 2022-06-30:
Hi,

Not sure if I got the execution right,

```
  g++ test_main.cpp test_lmpc.cpp -I./include -lCatch2
```

but the `test_lmpc` fails with the following error, any suggestions? Thanks

```
/usr/include/Eigen/src/Core/DenseStorage.h:33:3: error: static assertion failed: OBJECT_ALLOCATED_ON_STACK_IS_TOO_BIG
/usr/include/Eigen/src/Core/DenseStorage.h: In instantiation of ‘void Eigen::internal::check_static_allocation_size() [with T = double; int Size = 30976]’:
/usr/include/Eigen/src/Core/DenseStorage.h:110:41:   required from ‘Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 30976; int MatrixOrArrayOptions = 0]’
/usr/include/Eigen/src/Core/DenseStorage.h:187:38:   required from ‘Eigen::DenseStorage<T, Size, _Rows, _Cols, _Options>::DenseStorage() [with T = double; int Size = 30976; int _Rows = 176; int _Cols = 176; int _Options = 0]’
/usr/include/Eigen/src/Core/PlainObjectBase.h:484:55:   required from ‘Eigen::PlainObjectBase<Derived>::PlainObjectBase() [with Derived = Eigen::Matrix<double, 176, 176, 0, 176, 176>]’
/usr/include/Eigen/src/Core/Matrix.h:292:5:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::Matrix(const T&) [with T = Eigen::ReturnByValue<Eigen::KroneckerProduct<Eigen::Matrix<double, 11, 11, 0, 11, 11>, Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, const Eigen::Matrix<double, 16, 16, 0, 16, 16> > > >; _Scalar = double; int _Rows = 176; int _Cols = 176; int _Options = 0; int _MaxRows = 176; int _MaxCols = 176]’
/usr/include/Eigen/src/Core/DenseBase.h:406:48:   required from ‘Eigen::DenseBase<Derived>::EvalReturnType Eigen::DenseBase<Derived>::eval() const [with Derived = Eigen::ReturnByValue<Eigen::KroneckerProduct<Eigen::Matrix<double, 11, 11, 0, 11, 11>, Eigen::CwiseUnaryOp<Eigen::internal::scalar_opposite_op<double>, const Eigen::Matrix<double, 16, 16, 0, 16, 16> > > >; Eigen::DenseBase<Derived>::EvalReturnType = const Eigen::Matrix<double, 176, 176, 0, 176, 176>]’
/usr/local/include/mpc/ProblemBuilder.hpp:417:61:   required from ‘bool mpc::ProblemBuilder<Tnx, Tnu, Tndu, Tny, Tph, Tch>::buildTITerms() [with int Tnx = 12; int Tnu = 4; int Tndu = 4; int Tny = 12; int Tph = 10; int Tch = 10]’
/usr/local/include/mpc/ProblemBuilder.hpp:173:16:   required from ‘bool mpc::ProblemBuilder<Tnx, Tnu, Tndu, Tny, Tph, Tch>::setStateModel(mpc::mat<Tnx, Tnx>&, mpc::mat<Tnx, Tnu>&, mpc::mat<Tny, Tnx>&) [with int Tnx = 12; int Tnu = 4; int Tndu = 4; int Tny = 12; int Tph = 10; int Tch = 10; mpc::mat<Tnx, Tnx> = Eigen::Matrix<double, 12, 12, 0, 12, 12>; mpc::mat<Tnx, Tnu> = Eigen::Matrix<double, 12, 4, 0, 12, 4>; mpc::mat<Tny, Tnx> = Eigen::Matrix<double, 12, 12, 0, 12, 12>]’
/usr/local/include/mpc/LMPC.hpp:190:45:   required from ‘bool mpc::LMPC<Tnx, Tnu, Tndu, Tny, Tph, Tch>::setStateSpaceModel(mpc::mat<Tnx, Tnx>&, mpc::mat<Tnx, Tnu>&, mpc::mat<Tny, Tnx>&) [with int Tnx = 12; int Tnu = 4; int Tndu = 4; int Tny = 12; int Tph = 10; int Tch = 10; mpc::mat<Tnx, Tnx> = Eigen::Matrix<double, 12, 12, 0, 12, 12>; mpc::mat<Tnx, Tnu> = Eigen::Matrix<double, 12, 4, 0, 12, 4>; mpc::mat<Tny, Tnx> = Eigen::Matrix<double, 12, 12, 0, 12, 12>]’
test_lmpc.cpp:62:44:   required from here
```

nicolapiccinelli, 2022-06-30:
Look at the CMakeLists in the test folder as a reference. You are missing the flag disabling that Eigen assertion. Moreover, if you want to have all the best performance you should add -O3 and enable openmp

closed 2022-07-05 by nicolapiccinelli

abdelhakim96, 2023-10-26:
I get the same error. Can you specifically specify what part of CMakelists is should include to avoid such error?

nicolapiccinelli, 2023-10-26:
This is the definition to add, if you need further information about check this out [https://eigen.tuxfamily.org/dox/TopicPreprocessorDirectives.html](https://eigen.tuxfamily.org/dox/TopicPreprocessorDirectives.html)
```
# Disabling Eigen stack allocation warning
add_definitions(-DEIGEN_STACK_ALLOCATION_LIMIT=0)
```