nicolapiccinelli/libmpc #8: Future variables for NLMPC

Veng97, 2022-12-17:
For my task i  need to be able to specify online parameters in some way within the model itself. Specifically i need the ability to set the curvature for a cross-track-error model. This requires adding an additional horizon variable, e.g:

```
auto stateEq = [&](mpc::cvec<Tnx>& dx,
                   mpc::cvec<Tnx> x,
                   mpc::cvec<Tnu> u,
                  mpc::cvec<Tnv> v) {
...
};
```

The new variable, v, can be used for online parameter settings along the entire prediction horizon. This would also allow updating parameters with online estimates of disturbances or model parameters.

nicolapiccinelli, 2023-01-06:
Hello @Veng97 and happy new year. I just saw your request currently, I will give a look in detail next week. However, this is something that is currently not available, but for further expansion, it could be possible to provide the index of the current horizon step as a parameter of the function handles

nicolapiccinelli, 2023-01-07:
The solution mentioned above will be available in the next release, cheers!

nicolapiccinelli, 2023-01-09:
See release [0.2.0](https://github.com/nicolapiccinelli/libmpc/releases/tag/0.2.0)

closed 2023-01-09 by nicolapiccinelli