nicolapiccinelli/libmpc #9: Future references for NLMPC

Veng97, 2022-12-17:
The NLMPC doesn't seem to have a way of setting the future reference output. I guess something similar to the implementation in LMPC, that adds the ability to set a reference output, yr, and reference input, ur. Maybe we can add these two vectors directly in the definition of the ObjectiveFunction handle.

nicolapiccinelli, 2023-01-06:
@Veng97 I suggest moving this feature request together with issue #8 since the solution would be the same

nicolapiccinelli, 2023-01-06:
Hello @Veng97, in this case, if you give a look at the test `test_vanderpol.cpp` the `setObjectiveFunction` method provides the whole horizon for the input command and the state vector. So what you can do to have references is to capture some external matrices in your code and later use them in the lambda.

```
mpc::mat<TVAR(pred_hor + 1), TVAR(num_states)> x_r,
mpc::mat<TVAR(pred_hor + 1), TVAR(num_inputs)> u_r,

...

optsolver.setObjectiveFunction([&](
                                       mpc::mat<TVAR(pred_hor + 1), TVAR(num_states)> x,
                                       mpc::mat<TVAR(pred_hor + 1), TVAR(num_inputs)> u,
                                       double) {
        return (x_r.array() - x.array()).square().sum() + (u_r.array() - u.array()).square().sum();
    });
```

What is really missing here is the possibility of directly penalizing the system's output vector, which can be definitely added.

nicolapiccinelli, 2023-01-07:
@Veng97 I've added the output vector sequence in the lambda, once we completed your pr I will release a new version containing this

nicolapiccinelli, 2023-01-09:
See release [0.2.0](https://github.com/nicolapiccinelli/libmpc/releases/tag/0.2.0)

closed 2023-01-09 by nicolapiccinelli