nicolapiccinelli/libmpc #10: Discretization Method

makeller1, 2023-06-11:
I think it would be useful to inform what method is used for discretization. Looks like zero-order-hold. Would like to see that stated in the docs.

nicolapiccinelli, 2023-06-11:
Hello, @makeller1 to what discretization method are referring to? There are few routines used both in linear and nonlinear MPC.

Are you encountering issues due to the discretization of the system model or to the automatic gradient estimation? If so, it would be useful to describe better your use case.

Cheers!

makeller1, 2023-06-11:
Ideally a discretization method would be attributed to each routine. I do not know of any issues related to it, just that the discretization method informs design decisions around step size.

nicolapiccinelli, 2023-06-11:
In the linear mpc, the discretization is up to the user, there's just a util routine to compute the zoh. For the nonlinear case currently, if the user provides a continuous time system the solver internally converts it using zoh. Anyhow, even in this case, you can implement the discretization routine you prefer by directly providing the discrete system model.

I will put a note of that in the docs.

makeller1, 2023-06-11:
Great, thank you!

closed 2023-06-11 by makeller1