nicolapiccinelli/libmpc #12: Error building library

abdelhakim96, 2023-10-18:
OS: Ubuntu 20

I get the following error when building, in line 410 in LOptimizer.hpp

error: ‘csc’ has not been declared bool create OsqpSparseMatrix(const smat &eigenSparseMatrix, csc *&osqpSparseMatrix)

nicolapiccinelli, 2023-10-18:
@abdelhakim96 it is due to a change in the OSQP headers. To account for that you can use the `configure.sh` from the master branch. The configure script in the release versions <= 0.4.0 does not specify which OSQP version should configure the dependencies.

abdelhakim96, 2023-10-18:
Thanks alot! i run the configure script and now the error is gone. 

I get another error though, I am not sure if it is also related.

 In line 98 in IDimensionable.hpp

 error: taking address of rvalue [-fpermissive]    const mat<sizer.ph + 1, sizer.nu> &

nicolapiccinelli, 2023-10-18:
Regarding this error I think is due to your implementation. Please provide an MNWE to reproduce the error.

abdelhakim96, 2023-10-18:
I am building the project, without any implementation yet. My OS is Ubuntu 20

nicolapiccinelli, 2023-10-18:
Are building the tests? However please check if you are using the GCC version compatible with the library. Check the readme for that

abdelhakim96, 2023-10-18:
Yes Im building the tests. Im using Clion IDE which I think uses a different compiler than GCC, could this be the issue? Is there a way to build the tests?

abdelhakim96, 2023-10-18:
I have changed my compiler to g++ 10.5 and i still get the same error building the tests

nicolapiccinelli, 2023-10-18:
IDK what is going on in your system in the CI/CD environment it is working properly. Anyway, if you just need to use the library I suggest compiling it in the release mode like that

```
cmake -D CMAKE_BUILD_TYPE=Release -D CMAKE_EXPORT_COMPILE_COMMANDS=ON ..
make install
```
in this way, it will install the library in the system and you can use it in your project.

abdelhakim96, 2023-10-18:
It was an issue with the compiler as you have mentioned, when i updated to the recommended version it worked! Thank you very much for your help

closed 2023-10-18 by abdelhakim96