nicolapiccinelli/libmpc #13: error: static assertion failed: YOU_MIXED_MATRICES_OF_DIFFERENT_SIZES

abdelhakim96, 2023-10-21:
I am trying to replicate the van der pol example in the tests to understand how to use the library.

I get this error when using the setObjectiveFunction in NLMPC

error: static assertion failed: YOU_MIXED_MATRICES_OF_DIFFERENT_SIZES

here is a MWE to replicate the error:

#include <mpc/NLMPC.hpp>
#include <mpc/LMPC.hpp>

int VanderPol()
{
    constexpr int num_states = 2;
    constexpr int num_output = 2;
    constexpr int num_inputs = 1;
    constexpr int pred_hor = 10;
    constexpr int ctrl_hor = 5;
    constexpr int ineq_c = pred_hor + 1;
    constexpr int eq_c = 0;

    double ts = 0.1;

    mpc::NLMPC<
            num_states, num_inputs, num_output,
            pred_hor, ctrl_hor,
            ineq_c, eq_c>
            optsolver;

    optsolver.setLoggerLevel(mpc::Logger::log_level::NORMAL);
    optsolver.setContinuosTimeModel(ts);

    auto stateEq = [&](
        mpc::cvec<num_states> &dx,
        const mpc::cvec<num_states>& x,
        const mpc::cvec<num_inputs>& u)
    {
        dx(0) = ((1.0 - (x(1) * x(1))) * x(0)) - x(1) + u(0);
        dx(1) = x(0);
        dx(2) = x(0)-1;
    };

  optsolver.setStateSpaceFunction([&](
       mpc::cvec<num_states> &dx,
        const mpc::cvec<num_states>& x,
      const mpc::cvec<num_inputs>& u,
        const unsigned int&)
    { stateEq(dx, x, u); });

    optsolver.setObjectiveFunction([&](
        const mpc::mat<pred_hor , num_states> &x,
        const mpc::mat<pred_hor, num_output> &,
       const mpc::mat<pred_hor , num_inputs> &u,
        double)
   { return x.array().square().sum() + u.array().square().sum(); });

    optsolver.setIneqConFunction([&](
        mpc::cvec<ineq_c> &in_con,
        const mpc::mat<pred_hor + 1, num_states>&,
        const mpc::mat<pred_hor + 1, num_output>&,
        const mpc::mat<pred_hor + 1, num_inputs>& u,
        const double&)
    {
        for (int i = 0; i < ineq_c; i++) {
            in_con(i) = u(i, 0) - 0.5;
        } });

    mpc::cvec<num_states> modelX, modeldX;
    modelX.resize(num_states);
    modeldX.resize(num_states);

    modelX(0) = 0;
    modelX(1) = 1.0;

    auto r = optsolver.getLastResult();

    for (;;)
    {
        r = optsolver.step(modelX, r.cmd);
        auto seq = optsolver.getOptimalSequence();
        (void) seq;
        //stateEq(modeldX, modelX, r.cmd);
        modelX += modeldX * ts;
        if (std::fabs(modelX[0]) <= 1e-2 && std::fabs(modelX[1]) <= 1e-1)
        {
            break;
        }
    }

    return 0;
}

int main() {
    int result = VanderPol();
    if (result == 0) {
        std::cout << "VanderPol example completed successfully." << std::endl;
    } else {
        std::cerr << "VanderPol example encountered an error." << std::endl;
    }
    return result;
}

I would really appreciate your help.

Thanks!

nicolapiccinelli, 2023-10-23:
At first sight, there are a couple of things you have to fix:

1. do not add both the headers for linear and nonlinear mpc
2. in the stateEq you are trying to access the third element of the state vector, but the num_states is of 2 elements

abdelhakim96, 2023-10-23:
Thanks for your reply.
1. i have added it it similar to the tests by including basic.hpp, anyway, i have removed LMPC.
2. I removed the extra state but i still get the same error

would you have some ideas why this error might arise?

nicolapiccinelli, 2023-10-23:
`basic.hpp` is a header of the testing framework you just need the other one. Then, you should create a separate C++ project based on the sample CMake available in the README instead of putting your code directly in the libmpc++ project. However, if the tests work and you just copied their code in a different file I don't understand how you having such error.

abdelhakim96, 2023-10-25:
I still cant fix the isssue. Which eigen version should i be using?

abdelhakim96, 2023-10-25:
btw the test_cases_dynamic are the only ones that work for me, i get the same error when running the other 3 tests.

nicolapiccinelli, 2023-10-25:
If you clone the repo from the master and use the configure and build test it work? Can you report here the console output?

closed 2023-11-03 by nicolapiccinelli