nicolapiccinelli/libmpc #16: Does the library support real-time iteration mpc scheme?

abdelhakim96, 2023-10-27:
Does the libmpc support real-time iteration mpc scheme, like acados library for example? If not, is extending it to support RTI possible?

nicolapiccinelli, 2023-10-27:
What does it means real time iteration?

abdelhakim96, 2023-10-27:
"The scheme allows a fast reaction to disturbances by delivering approximations of the exact optimal feedback control which are iteratively refined during the runtime of the controlled process."

This method is applied in acados mpc library https://docs.acados.org/about_acados/index.html

reference to method:
https://epubs.siam.org/doi/abs/10.1137/S0363012902400713?journalCode=sjcodc

nicolapiccinelli, 2023-11-03:
Hello @abdelhakim96, the library uses SLSQP to solve the underlying non-linear optimization problem. However, it is possible to plan to extend the non-linear optimizer class and include such a new scheme.

closed 2023-11-03 by nicolapiccinelli