nicolapiccinelli/libmpc #20: installing the library on ubuntu, documentation

marvin-ad-martianum, 2024-01-24:
Amazing library! Thank you. But i was struggling with installing it for a bit. However, i found this sequence of code working amazingly well. 

sudo ./configure.sh 
make
sudo make install

Now im aware that this is quite common, however if you add something like this to the readme in a way or another im sure you could make this code more easily accessible to less professional coders. This package is great! Im from ETH zurich and use it for a ros2 implementation.

nicolapiccinelli, 2024-01-24:
@marvin-ad-martianum great to hear you are enjoying the lib. IIf you want to contribute open a pull request and update the readme accordingly. Would be very appreciated!

nicolapiccinelli, 2024-01-30:
@marvin-ad-martianum, see 0.4.1 for the updated `configure.sh` script

closed 2024-01-30 by nicolapiccinelli