nicolapiccinelli/libmpc #22: What is the license?

AndyZe, 2024-01-29:
I'd like to use this. It looks well-done. But, it will need some type of open-source license.  :+1:

nicolapiccinelli, 2024-01-29:
@AndyZe I'm not really an expert on licencing open-source software. Do you have suggestions?

AndyZe, 2024-01-30:
The YOLO/Darknet [license](https://github.com/AlexeyAB/darknet/blob/master/LICENSE) is my favorite:

0. Darknet is public domain.
1. Do whatever you want with it.
2. Stop emailing me about it!

Most robotics software is using Apache 2.0 these days.

nicolapiccinelli, 2024-01-31:
I just opened a pull request for the license. The license will be MIT since the code base is header only. I don’t see any particular problem with the other dependencies. So mostly like Darknet.

closed 2024-01-31 by nicolapiccinelli