nicolapiccinelli/libmpc #24: Use with Unicycle Model

nocholasrift, 2024-02-12:
Greetings! Great package, looks very interesting. I'm curious to know how easy it would be to setup this MPC to work with a unicycle model platform. Have you all tried with such a model? If so, was the MPC able to run at 10-15Hz? Is any linearization required?

Thanks!

nicolapiccinelli, 2024-02-12:
Hello @nocholasrift, I’ve published a paper about motion planning of mobile robots using the library: https://ieeexplore.ieee.org/abstract/document/10178219

The implementation I did was using the non-linear MPC, and I was able to keep around 10/15hz. Of course, it depends on the number of constraints, the computation effort required, etc.

nocholasrift, 2024-02-12:
Thanks for the follow up, very helpful! Out of curiosity, is the source code for that non-linear mpc formulation with the gradients available publicly anywhere, or is it private?

nicolapiccinelli, 2024-02-12:
The gradients in the non-linear MPC are automatically established using finite differences. Regarding the implementation, I have it on my private repository. If you think it would make sense, I can add a simplified version as an example.

nocholasrift, 2024-02-12:
No worries, you have a pretty clear example of the NLMPC setup in the test_vanderpol.cpp file, so I can go from there! I am curious though. From the NLOPT documentation: "If you find yourself computing the gradient by a finite-difference approximation (e.g. ∂f/∂x≈[f(x+Δx)−f(x−Δx)]/2Δx in one dimension), then you should probably use a derivative-free algorithm instead." I realize this library is developed as a general use MPC so it would not be feasible to implement in that way, but have you played around at all with using analytical gradients in the MPC formulations / noticed any speedups from doing so?

nocholasrift, 2024-02-12:
Gotcha, I'm surprised that is the case, but makes sense. SLSQP definitely seems like the way to go, from my own experience, none of the other algorithms compare in this use-case. Thanks for the discussion and quick follow-ups!

closed 2024-02-12 by nocholasrift