nicolapiccinelli/libmpc #26: Can you provide ROS + pluginlib example?

sm3304love, 2024-03-19:
Hello! I am a student trying to implement motion planning using NMPC. The libmpc library looks appealing, but I am unable to use it because the C++ versions between ROS and libmpc do not match. I saw an answer to a similar question that suggested using Pluginlib, but unfortunately, I do not know how to import a library into a code using Pluginlib. Could you please share an example of using libmpc in ROS through Pluginlib?

nicolapiccinelli, 2024-03-19:
Hello @sm3304love, can you provide some information about your setup? ROS version, OS, compiler version etc.

sm3304love, 2024-03-20:
> Hello @sm3304love, can you provide some information about your setup? ROS version, OS, compiler version etc.

Yes. My setup information is as follows.

- ROS: Noetic
- OS: Ubuntu 20.04
- gcc version: 9.4.0

If you need any more information, please let me know!

nicolapiccinelli, 2024-03-20:
I will send you something soon

sm3304love, 2024-03-26:
Hello, may I know when I could receive the example? I greatly appreciate your assistance.

sm3304love, 2024-03-26:
> I will send you something soon

Thank you! My email address is in my github profile if you need it

nicolapiccinelli, 2024-03-26:
Check your inbox

closed 2024-03-26 by nicolapiccinelli