nicolapiccinelli/libmpc #27: How to set input, state constraints using NLMPC?

sm3304love, 2024-03-29:
Hello! You kindly answered my question last time, so I was able to successfully apply libmpc to ROS.

I have a new question from implementing NMPC.

The example using LMPC showed that it can easily set min and max constraints for the state, input, and output using the **setConstraints** function, but this function does not appear to be available for NLMPC. How do I set min and max constraints for state, input, and output in NLMPC?

nicolapiccinelli, 2024-03-29:
It's good to hear you successfully integrated the controller in ROS. Regarding the definition of box constraints, you can look to the vanderpol or the UGV example.

Differently from the linear MPC, in this case the constraints functions are user defined, so if you simply need bounds you can iterate over the horizon and set the bounds accordingly.

Please refer to the manual where you can find details about the api and much more

sm3304love, 2024-04-01:
Referring to the API and example, I set inequality constraints for input and state as follows and confirmed that it worked well.
```
mpc_solver.setIneqConFunction([&](mpc::cvec<ineq_c> &in_con, const mpc::mat<pred_hor + 1, num_states> &x,
                                      const mpc::mat<pred_hor + 1, num_outputs> &,
                                      const mpc::mat<pred_hor + 1, num_inputs> &u, const double &) {
        Eigen::VectorXd x_lb = (Eigen::VectorXd(12) << joint_pose_lower_limit, joint_vel_lower_limit).finished();
        Eigen::VectorXd x_Ub = (Eigen::VectorXd(12) << joint_pose_upper_limit, joint_vel_upper_limit).finished();

        int index = 0;

        for (int i = 0; i < pred_hor + 1; i++)
        {
            for (size_t j = 0; j < num_inputs; j++)
            {
                in_con(index++) = u(i, j) - 1.0;  // u <= u_max
                in_con(index++) = -u(i, j) - 1.0; // u_min <= u
            }
            for (int j = 0; j < num_states; j++)
            {
                in_con(index++) = x(i, j) - x_Ub(j);  // q <= q_max
                in_con(index++) = -x(i, j) - x_lb(j); // q_min <= q
            }
        }
    });
```

closed 2024-04-01 by sm3304love