nicolapiccinelli/libmpc #32: Are there any plans to support other NLP solvers?

sm3304love, 2024-04-24:
I am trying to implement motion planning where the mobile manipulator avoids obstacles based on nmpc. In general, if there are no obstacles, planning speed is good(approximately 0.03 sec).  However, if an obstacle approaches the robot (corresponding to the obstacle avoidance constraint), a solution is found, but the duration is very slow (approximately 0.4 seconds) or no solution is found. I tried changing the SLSQP solver in NLOptimizer.hpp to another solver, but the other solver does not work well as shown in the image below.

When using the LD_VAR1 algorithm
[image]

Additionally, since SLSQP, currently used in libmpc, is suitable for problems with few variables, it is expected that performance will deteriorate as the horizon becomes longer and inequality constraints increase. The Interior Point method is generally said to have better convergence speed and solution finding ability than the SLSQP method in complex problem. Do you have any plans to support Ipopt?

sm3304love, 2024-04-24:
The number of constraints for each step is as follows.

Number of States: 9
Number of inputs: 9
Number of Inequality constraints per Horizon: 54 (state, input bound + obstacle avoidance + self collision avoidance)

nicolapiccinelli, 2024-05-07:
SLSQP should scale quite well up to some thousands of constraints. You can also try to tune the convergence tolerances and ensure you are compiling the code with the optimisation flags enabled.

Anyhow, the support for another solver than NLOpt is not in view. But could be interesting to give the hooks for an external solver, so the user if they need can integrate another solver. But it should be carefully considered.

closed 2024-05-20 by sm3304love