nicolapiccinelli/libmpc #35: NLMPC State Bounds Exceeded

tchase24, 2024-08-22:
I'm running into an issue when using the NLMPC with state bounds. I would expect that specifying a min/max along the prediction horizon would narrow the search space and remove instances where the bounds are exceeded. However, instead I get a solver error of `Internal solver error: bounds 14 fail 0 <= -0.19834 <= 2`. Is this the behavior that is intended?

nicolapiccinelli, 2024-08-24:
@tchase24, can you provide more insight about your platform, library version, compiler version and dependencies version? Do you verify if your optimization problem is initialized in a feasible initial condition?

tchase24, 2024-08-26:
@nicolapiccinelli it was an initial condition problem. The original state was fine but the propagation allowed it to go out of bounds. Appreciate the assistance.