nicolapiccinelli/libmpc #36: Custom numerical integration for MPC predictions

sara9915, 2024-09-17:
Hello @nicolapiccinelli, I'd like to use your library, but I need to employ a custom numerical integrator to make the MPC predictions along the prediction horizon. Can you give me some info about the routine now used to do this? I see there is an RK4 class, but I didn't find anywhere its usage in the provided examples.

nicolapiccinelli, 2024-09-17:
Hello, @sara9915. Are you referring to the integration of the model dynamics?

sara9915, 2024-09-17:
Yes, exactly

nicolapiccinelli, 2024-09-19:
Ok, the library is handling things differently between linear and non-linear MPC. Which one are you interested in?

sara9915, 2024-09-19:
Non-linear MPC

nicolapiccinelli, 2024-09-19:
If you already have a discrete-time model, the library is simply propagating the state x(k+1) = f(x(k),u(k)). Otherwise, if you provide a continuous time model, the integration is performed using the sample time's central difference.  In this case, the integration exploits the prediction horizon to implement the integration scheme.

sara9915, 2024-09-20:
I got it, thank you!

closed 2024-09-20 by sara9915